bambulab/Bambu-Handy #1: Notifications go to only one device, and there is no way to change which device that is.

mapaolini, 2022-09-01:
How can you change which device the Printer is sending notices to?

More specifically, I have an IPhone logged into Bambu Handy (which I used first with the printer, and to which I’d like to have the notifications go to), and I have an iPad logged in with Bambu Handy 9was the second device).  The notifications always go to the iPad, even when it’s not being used instead of the iphone (and watch). 

Ideally printer notifications would go to all devices logged into the account and connected to the printer, but I’ll take rerouting to the phone for now. 🙂

(Guessing I may have to log out from the iPhone and log back in, but I’d thought I’d ask here first).

LibaoBambu, 2022-09-05:
Thank you for filing this issue! Only the latest login device will receive the notifications. If you need to receive notifications on your phone, you only need to restart bambu handy to update your deviceToken.

mapaolini, 2022-09-05:
I've restarted the Bambu Handy App on my phone several times without it changing to the primary notification device.  Or do you mean log out and log back in?

mapaolini, 2022-09-09:
I've now logged out of Bambu Handy, and back into Bambu Handy on my iPhone.  Did this without opening Bambu Handy on my iPad.   Notifications are still going to my iPad...

JstBecause, 2022-09-22:
Yes please. Would like to get notifications on all devices that have the app installed.

I've tried logging out of all devices, logging back into just the app on my iPhone. And still no notifications.

The girlfriends android though, gets all the notifications.

joviannm, 2023-01-27:
Would also really like to be able to get push notifications on multiple devices. Both my partner and I run the printer and would like to both get push notifications.

renssies, 2023-03-15:
I consider this a serious oversight by the team. Any notifications for the Bambu handy app are only received on the device that last used the app.

Well I use both an iOS and Android device throughout the day and I also have an significant other who is using the X1 Carbon.

Now it's guess work where the notification arrives.

The app should register the token for every single device that has logged in to the app, and remove it on log out.

If anyone advice is needed by the team, I'm senior mobile developer and happy to help :)

renssies, 2023-04-13:
Hi @LibaoBambu,

You are likely responding to the wrong email, this issue is about the notifications of the Bambu Handy app, not the AMS.

LibaoBambu, 2023-04-13:
> Dear Ryan Eeen
>
> I found the wiki article you mentioned,  https://wiki.bambulab.com/en/p1/manual/connect-ams, and it actually does have information on connecting to AMS. If your issue remains unresolved, please do not hesitate to reach out to us.
>
> Best regards.

I'm so sorry to intrude on you, please ignore this email.

LibaoBambu, 2023-04-13:
I am sorry for the mistake I madein previous email.

Currently, about push notificationissue and I will let you know once the problem is solved. I apologize for any inconvenience this may have caused.

Thank you for your patience and understanding.

LibaoBambu, 2023-05-12:
The feature Multi-device receive push has been released.  And many features have been included in this new version update https://github.com/bambulab/Bambu-Handy/releases/tag/1.1.0 Please pay attention to updates in the App Store or Google Play.
If there are any questions during the usage process, please let me know. thanks.

LibaoBambu, 2023-12-05:
Feature has been supported.

closed 2023-12-05 by LibaoBambu