bambulab/Bambu-Handy #2: Printer offline unless you reload the app iOS 15.6.1

CNCXPROUK, 2022-09-04:
Bambu handy often says printer offline when printing unless you end the app and reload it on iOS 15.6.1.

pepsiliu, 2022-09-05:
@CNCXPROUK Please provide the handy version.

CNCXPROUK, 2022-09-05:
1.0.3

pepsiliu, 2022-09-05:
@CNCXPROUK Thanks for ur report. We have fix an offline issue caused by unstable network environment or turn off/on the printer manually during using handy. We'll release next version very soon.

CNCXPROUK, 2022-09-05:
At the moment that's not practical when the printer is printing. My network is stable, it could be because I have a mesh network?

pepsiliu, 2022-09-06:
Let us double check about it. Is there any more details to update?

CNCXPROUK, 2022-09-06:
Not much. It appears to happen if I haven’t launched the app to check on a print for a while.  I am not sure if it’s because iOS is suspending the app and when you re open it, it doesn’t initiate the connection the same way as when you open the app after a reboot.

Tomtom5893, 2023-03-24:
Aame on Android.

alaaneme, 2024-05-19:
This is still happening, I can replicate the problem very easily. iPhone 12 pro latest iOS, Bamby Handy 2.10 (latest). Very annoying when I quickly want to check on the print progress

exedriver, 2024-05-28:
Same for Bambu Handy 2.10 on Android (Samsung Galaxy S20FE, latest OS)