bambulab/Bambu-Handy #5: iOS Dark mode

dereknheiley, 2022-09-09:
iOS handy app should support native dark mode (os setting) or manual mode, including the loading screen, for example like Google chrome

[image]
[image]
[image]
[image]

JustinRudebaugh, 2022-10-20:
100% agreed!

DoubleTts, 2023-02-22:
yes, please

MrDaGree, 2023-02-25:
+1 would love to have this feature! Theres been times especially at night where im not excited about blinding myself while checking on a print

pepsiliu, 2023-05-17:
Released on 1.1.0

closed 2023-05-17 by pepsiliu