bambulab/Bambu-Handy #9: Android Permission Issues

Podunk209, 2022-09-28:
Get "Need permission to access your photos" error message, even though permissions are set.  Tried deleting storage and cache and re-installing app and still not working.  Version 1.0.5 on Pixel 6 with Android 13.

pepsiliu, 2022-10-10:
> When trying to attach a photo or video in the ticket submission section I get "Need permission to access your photos" error message, even though permissions are set. Tried deleting storage and cache and re-installing app and still not working. Version 1.0.5 on Pixel 6 with Android 13.

I have tested on Handy (1.0.5), it works fine for me. Maybe u didn't allow handy to access 'Files and media' permission.

Podunk209, 2022-10-11:
I'm running 1.0.5 as well.  Under App permissions I don't see an option to allow 'Files and media'  I've allowed every requested permission and show no permissions under "Not allowed" (see screenshot).
[image]

pepsiliu, 2022-10-12:
Handy permission check fail because It didn't adapt storage permission on Android 13. It will be fixed on next release version.

pepsiliu, 2022-12-15:
View on version 1.0.6

closed 2022-12-15 by pepsiliu