bambulab/Bambu-Handy #12: Notification for filament runout

VACIndustries, 2022-10-02:
We should get a or be able to enable a notification for filament runout

pepsiliu, 2022-10-08:
Any Pause/Stop event from the printer will trigger a notification to app at present .

DeanDavis, 2022-10-24:
I do not get iOS notification for any event. I’ve double checked everything. I have an entry in the notifications section of my settings app for Bambu Handy, everything is turned on but if I get a filament runout for instance I don’t know it until I open the app or physically check the printer.

pepsiliu, 2022-11-01:
> I do not get iOS notification for any event. I’ve double checked everything. I have an entry in the notifications section of my settings app for Bambu Handy, everything is turned on but if I get a filament runout for instance I don’t know it until I open the app or physically check the printer.

Could u please provide ur account email to help us to check this issue.

DeanDavis, 2022-11-01:
afterten@gmail.com

pepsiliu, 2022-11-04:
> [afterten@gmail.com](mailto:afterten@gmail.com)

We have checked the events send to this account, there is no event cause of a filament runout. It seems the printer didn't send this event out. I suggest that you can submit a ticket when you meet this issue next time, which contains the printer logs.

Hologos, 2023-08-20:
I still don't get any notification on iOS if there is a problem with a print unless I open the app and check manually.

Hologos, 2023-08-20:
@pepsiliu I still to this day don't get any notification about print problems on iOS. I only receive print cancelled and print finished notifications.

ercdvs, 2023-08-27:
Bumping this issue, as its exactly the same on Android OS.   I ONLY get notifications on print cancelled or print complete.

Errors that have paused the print, including AMS filament runout,  The hotend cover coming off, z home failing ,etc have never been sent to my phone and are only viewable via bambu studio.   This is the basic functionality of what we use this monitoring for.

If you want to run everything through the bambu ecosystem, then provide all the notifications.  Alternately provide access to 3rd party systems to notify users.

relax81, 2023-11-02:
I had the same issue today. Got my Bambu P1S just a few days ago and noticed that the Filament Runout Notification only shows as a red text line in the Bambu Studio app, but neither the Bambu Display or my Bambu Handy App showed me a filament runout message.

relax81, 2023-11-04:
Another day and I had the situation again. I've opened a ticket about it yesterday and added logs and information since then. 
Ticket No: US231102877001

LibaoBambu, 2023-12-05:
Feature has been supported.

Hologos, 2023-12-05:
@LibaoBambu If it's supported why it doesn't work for so many people? I just checked and it still doesn't work. Do you think it's a good idea to close this issue? What kind of support is that closing an unresolved issue?