bambulab/Bambu-Handy #26: QR code shown on printer screen should always work.

djunod, 2022-11-14:
When a QR code is shown on the printer screen for reporting a problem, it should work with the Handy app instead of saying you need an update (all firmware, Bambu studio & Handy code is up-to-date).

Attached is the QR code in question.

[image]

KD7LRJ, 2022-11-15:
For reference, here's what I see when trying to use this QR code on my phone (which is running Bambu Handy v1.0.5):

[image]

djunod, 2022-11-15:
I had finished a print, and selected "Report" when it was done.  That's when that QR code was shown.

pepsiliu, 2022-11-16:
It should be work, when user scan QR code only by the Handy.

pepsiliu, 2023-01-16:
Close it, since there is no more informations.
You can reopen it if this issue occurs again.

djunod, 2023-01-16:
> Close it, since there is no more informations.
> You can reopen it if this issue occurs again.

Did you try scanning the above QR code using Handy.  It still shows exactly the same thing, "Unsupported New Feature".  Definitely not fixed.

pepsiliu, 2023-01-16:
Waiting for cloud sync. I will make a double check about this issue if cloud is ready.