bambulab/Bambu-Handy #27: [Feature Request] Ability to upload to printer without printing.

BambuLabDiscord, 2022-11-17:
Slicer could simply add it to your cloud/history since that's where they go anyway after printing. Perhaps a new tab/page for queued. This would allow us to slice up multiple items ahead of time. I currently do this on my OctoPi's on my 6 Vypers and it's super handy.

Merlwynd, 2022-11-17:
Yes, I would also like to see this. I am trying to find a way to load a print file to the printer from Bambu Studio and then start the print from Bambu Handy at a later time…but have not found a way to do it.

AbeFM, 2022-12-04:
I came here today specifically to post this. I typically queue up several prints (sometimes in a single project file on different plates), then have to turn my computer back on (or travel to my computer, THEN turn it back on) when I already sliced the file long ago.

This would highly increase the value of the system to any sort of farm activity.

pepsiliu, 2022-12-15:
We are working on it. Users now can send files to printer via latest Bambu Studio

AbeFM, 2022-12-15:
Oh wow, big update. Super excited to check this out! Bambu's responsiveness is pretty awesome.

pepsiliu, 2022-12-22:
> I have problems with sending to printer I got an error that printer is not in same network but is. [image]

Have forwarded to Studio.

closed 2023-12-05 by LibaoBambu