bambulab/Bambu-Handy #36: [Feature Request] Live Activity Support

BambuLabDiscord, 2022-12-01:
add a live activity for ios that allows the printers status from the lock screen or from the dynamic island.

pepsiliu, 2022-12-22:
That's a very good interaction, we'll take it to our feature list.

freeformz, 2023-07-12:
+1 https://forum.bambulab.com/t/feature-request-support-live-activities-on-ios/19747

Josh4532, 2023-07-12:
+1 this would be cool

stef1949, 2023-12-07:
+1 any update for this?

stef1949, 2024-02-16:
?

cfuller007, 2024-02-17:
I would love the same along with support to Dynamic Island. I just got my first Bambu lab printer and I really missed that from OctoPod.

freeformz, 2024-05-07:
I started using Joint Craft for this: https://apps.apple.com/us/app/joint-craft-3d-print-monitor/id6455888249

stef1949, 2024-08-01:
> That's a very good interaction, we'll take it to our feature list.

Any update on this?

Alshootfa, 2024-10-21:
+1 from me too. This feature is probably the most needed feature for Bambu Handy in my point of view.

MilkMGN, 2024-10-22:
+1 Something Octopod looking would be nice. Even an Apple watch inrtegration too

StevenLi-phoenix, 2025-03-02:
@BambuLabDiscord @pepsiliu 
This issue could be closed now. It is implemented in 2.19.0
https://apps.apple.com/us/app/bambu-handy/id1625671285
```
- Added support for Live Activities to show print progress on ios
- 增加了对 Live Activities 的支持，以在 iOS 上显示打印进度
```

closed 2025-03-04 by LibaoBambu

StevenLi-phoenix, 2025-03-08:
seems hot fix broke it again ...

stef1949, 2025-03-16:
@LibaoBambu 💖

StevenLi-phoenix, 2025-03-17:
> stef1949
>  left a comment
> (bambulab/Bambu-Handy#36)
> @LibaoBambu 💖

Live activities only works if you open the app while printing.
And it works now
Thank you devs

AndaleR, 2025-06-10:
I was looking for that a long time. And then I found this point and I needed to activate it. And restart the app. NOW it works.[image]