bambulab/Bambu-Handy #39: [Feature Request]

brianiup, 2022-12-03:
When the printer encounters an error like AMS motor overload because a spool is stuck or similar issue please send us a notification to the Handy push notification so we can correct the issue. Please allow for notifications to be customizable in the sense you can enable or disable this notification.

I have had multiple stops due to an AMS motor overload due to a stuck spool, but never received any notification of the issue except for noticing the printer stopped when checking in on it.

AbeFM, 2022-12-04:
+1. If the printer stops for any reason, I'd like real time feedback.

DoubleTts, 2023-02-22:
+2, I have a pause set in some my prints and it would be nice to know it