bambulab/Bambu-Handy #40: Could we have a "hardware" reset button?

AbeFM, 2022-12-04:
1-3 times a week, the printer will stop accepting uploads, or have some other issue requiring a reboot.

I have to shuffle around my workspace, reach the off switch, wait... I don't know how long... and eventually turn it back on.

I would like the red arrow'ed Reset button on the top of the printer to reset the machine if you hold it for 7 seconds, or press it exactly 4 times, or something similar. Even a pop-up on the screen would be ok.

Is this at all possible?

sodachen22, 2023-01-09:
@AbeFM Thank you for the suggestion. But I want to confirm when the printer requires a reboot, and in your case, can the reboot resolve the issue?
What do you want is a software reboot or a really power off?

AbeFM, 2023-01-11:
That I'm unsure of.
The only method I've had available is a power cycle. While not a major concern, it can be more damaging to equipment to turn the mains on and off. I'd worry if done too quickly you might have bunk information in memory that could cause the machine to misbehave as well.

My feeling would be send a hardware reset pulse to all boards and peripherals, to essentially get the machine to do what it would do if you properly cycled the power.

I'm well open to more discussion on the idea. My main point is the button doesn't do a whole lot and I find myself cycling the machines (when I'm having some issue, often networking issue where many prints fail to send but including mechanical jams, etc) by means of the switch and I'd like to save both the effort and the wear (and the time of counting out 10? 15? 30 seconds?).

I'm unsure of what is available to ask for. :-) Thanks again!

werewolfenn, 2023-01-15:
I like this idea, but would really like a similar reset / restart button in the Handy app. If I'm not in the same room or building, but a simple reset could fix something without having the delay of getting back to the actual machine, that would be great.

sodachen22, 2023-01-16:
@AbeFM In our testing,  power cycle does not have damage to the printer. In your case, we may need a button for power cycle, we can have consider this in the next product.

AbeFM, 2023-03-19:
Picking this thread back up, 

> @AbeFM Thank you for the suggestion. But I want to confirm when the printer requires a reboot, and in your case, can the reboot resolve the issue?
> What do you want is a software reboot or a really power off?

That was the heart of the matter and I got distracted. You make a great point: I generally have two issues:
1) Sending prints over and over and they don't print
2) Printer not realizing a print has been canceled. I keep pressing stop and it's sitting there.

In both cases, it's the .... high level manager board, and/or the networking (maybe the same board) that appear to be stuck. Maybe it's a coms issue between the screen. 

The point is, what I really want is to unstick the machine without a switch flip. So:
**Can we have the red circle arrow "Reset" button actually reset the main controller board in a way which would recover from errors similar to those described above?**

Thanks for the insightful question, sorry it took me so long to realize it.