bambulab/Bambu-Handy #44: Humidity readout too small unreadable

Printman76, 2022-12-15:
1) The humidity readout is way too small and unreadable on even a large phone (I have a one plus 8 pro)

2) The readout should be a percentage not a fill meter
[image]

3) readout does not match the slicer
[image]

darkorb, 2022-12-17:
The humidity isn't percentage based, so the icon only changes color based on a few preset targets. The reasons are outlined in the release notes (and the data in the MQTT stream reflects this being like 0,1,2 etc. now and not 0-100).

pepsiliu, 2022-12-22:
Good suggestion. It will be optimized in the next version.

pepsiliu, 2023-04-24:
Updated on 1.0.14

closed 2023-04-24 by pepsiliu