bambulab/Bambu-Handy #46: [Feature Request] Printing Progress with Live Activities on iOS 16.1+

Jakubnim, 2022-12-17:
Hello, 

Is there any way that Bambu iOS developers implement Live Activities for iOS 16.1 and up devices + iPhone 14 Pro Dynamic island version? Would be nice to check from time to time progress on my iPhone without unlocking it, just a quick glance. 

[image]

here’s developer note from Apple how to implement that with short description: 

https://developer.apple.com/design/human-interface-guidelines/components/system-experiences/live-activities

It would be awesome addition, I use it with tracking and taxi apps, would be great to have it in Handy also.

Jakubnim, 2022-12-17:
Here’s more technical specification with activity kit which should enable Live tracking for print progress.

Sir-Gavin, 2022-12-17:
I think this would by HANDY 🤪

pepsiliu, 2022-12-22:
Duplicated with #36

closed 2022-12-22 by pepsiliu