bambulab/Bambu-Handy #49: [issue] no ios notifications

BambuLabDiscord, 2022-12-20:
Ok I’m confused.  My handy app doesn’t give any notifications on my iPhone.  I have it set correctly in iPhone notifications and everything.  Anyone else having this issue or know how to resolve?  Tried firmware updates, app updates, removing app and reinstalling - still no print notifications….  i either have to keep opening and chrcking the app to see progress / status or look on printer

pepsiliu, 2022-12-21:
Please provide more information, such as account email, printer dev_id. It would be more helpful if you add a ticket to give us the printer's logs.

Snortimer, 2023-04-23:
I'm having the same problem. It was giving me notifications up until a couple weeks ago.

LibaoBambu, 2023-12-05:
Please reopen if this issue shows up again.

closed 2023-12-05 by LibaoBambu