bambulab/Bambu-Handy #50: [Feature Request:
Full height prime / purge tower] Full height prime / purge tower

BambuLabDiscord, 2022-12-22:
Add a tick box so it is possible to have a prime / purge tall produce to the same height as the model. 
The reason behind this is I have noticed when printing with Silk PLA there is a colour difference once the prime / purge tower has reached its required height. Currently, I have to add an additional model of the same height to resolve the issue. 

I can provide images of the issues if required?

pepsiliu, 2022-12-23:
I think it's a feature for Bambu Studio?

lanewei120, 2023-01-03:
@BambuLabDiscord
to better trace this issue, could you create this issue under https://github.com/bambulab/BambuStudio?

thanks

pepsiliu, 2023-01-03:
View  [https://github.com/bambulab/BambuStudio](BambuStudio)

closed 2023-01-03 by pepsiliu