bambulab/Bambu-Handy #51: Live camera doesn't work with P1P

akbyrd, 2022-12-22:
I've just added the camera to my P1P. The camera icon shows up on the LCD screen and in Bambu Studio. However, pressing the play button in Bambu Handy to view the camera always fails with the error message "The printer cannot be found on the LAN".

I can see and control all other aspects of the printer with Bambu Handy, only the camera is not working.

Bambu Handy: 1.0.6
Firmware: 01.01.00.00

pepsiliu, 2022-12-23:
As the message said, P1P only supports liveview on the LAN currently. Keep the P1P and Handy on the same LAN network.

akbyrd, 2022-12-23:
I'm on the same LAN. I even disabled my 5 GHz to make sure both were on 2.4 GHz. I also just tried disabling my mobile network to ensure wifi is being used.

pepsiliu, 2022-12-23:
It seems like a problem about the router. Could you try it again, with the printer connected to mobile's hotspot?

akbyrd, 2022-12-23:
It's a Cox router with completely default settings. Bambu Studio can see the camera just fine.

pepsiliu, 2022-12-23:
"The printer cannot be found on the LAN" shows up, because Handy cannot find P1P by SSDP. Was the Studio connected at the same Wi-Fi?

akbyrd, 2022-12-23:
Yes. I just tested with my PC connected to the 5 GHz wifi and physically unplugged the ethernet cable. Bambu Studio is still able to view the camera.

There appear to be other users having issues with the camera in Handy as well, so I don't think this is entirely a unique issue.

pepsiliu, 2022-12-23:
Yes, other users may also have same issue, P1P' liveview on WAN is on our future feature list, our engineer are working on it.

When you changed the network, I suggest u to restart the Studio, It might cache the IP address of P1P which obtained from network cable. Or try it on Handy with  the printer connected to mobile's hotspot.

akbyrd, 2022-12-23:
I need it to work on LAN, not WAN.

I tried restarting Studio and the camera continued to work. I restarted my PC and the camera continued to work.

I used my mobile hotspot and Handy was able to use the camera.

So it appears to be wifi related, but only affects Handy, not Studio.

pepsiliu, 2022-12-23:
We'll make a double check on the different between Studio and Handy,  and discuss a more reliable resolution for this issue. And Thank you for your report.

akbyrd, 2022-12-23:
Thanks! Let me know if there are any additional diagnostics I can run to help.

akbyrd, 2022-12-23:
Just wanted to document some cases from today of others with a similar issue:

> I too cant seem to get the cam to work on the Bambu Handy app. Worked on windows Bambu slicer tho.
https://old.reddit.com/r/BambuLab/comments/zt42n2/cant_get_p1p_early_bird_camera_to_work/j1c4772/

> The camera view now works in Studio (Windows) but not in the android app. Still no idea about the files.
https://old.reddit.com/r/BambuLab/comments/zt40we/p1p_camera_and_media/j1byq52/

> Mobile. Desktop works fine for me.
https://old.reddit.com/r/BambuLab/comments/zt5bq3/p1p_camera_not_available_in_the_bambu_app/j1c8tus/

> Got the camera to read on bambu studio instead of handy.
https://discord.com/channels/970163353542606888/1041379047399227493/1055359608564236319

A couple of reactions here indicating camera working in Studio but not Handy
https://discord.com/channels/970163353542606888/1041379047399227493/1055669071263322133

> Apparently the camera is failing on the app. but it works on the slicer
https://discord.com/channels/970163353542606888/1041379047399227493/1055666091701059594

> try it in bambu lab studio. the app is not working for me
> if you can see the camera icon in the printer screen it's connected correctly
> I opened a ticket for camera failing in iOS app.
https://discord.com/channels/970163353542606888/1041379047399227493/1055641011893829633
https://discord.com/channels/970163353542606888/1041379047399227493/1055643461832945664

pepsiliu, 2023-01-04:
Found a possible reason which may cause this issue. 
Some Android mobiles need to accquire the wifi's multicast lock to received the device's notify when use upnp protocol.
It'll be fixed on next release version soon.

closed 2024-03-12 by pepsiliu