bambulab/Bambu-Handy #52: Fan controls on the P1P is buggy

akbyrd, 2022-12-22:
When I change the fan speed the UI in Bambu Handy jumps around randomly. Say the fan is at 100%. I press the minus button 5 times to set it to 50%. The UI will update randomly a few times and settle on some value other than 50%. For example, I might see it do this:
100% > 90% > 80% > 70% > 60% > 50% > 90% > 70%> 80%
and then sit there stuck on 80%. The physical fan seems to be set to 50% as expected.

When I turn the fan off I can also see this buggy behavior. The fan will show as 0% > 20% > 0%. It seems like the controls are attempting to show both "desired fan speed" and "actual fan speed" and the two values are clobbering each other.

This applies to both the part fan and the auxiliary fan.

Also, when I turn the fan off and back on it gets reset to 100% fan speed, instead of remembering the previous speed.

pepsiliu, 2022-12-23:
A known bug from P1P, It'll be fixed on next P1P's release version.

LibaoBambu, 2023-12-05:
Please reopen if this issue shows up again.

closed 2023-12-05 by LibaoBambu