bambulab/Bambu-Handy #53: [and can you please add raft air gap to your bambu slicer please and thank you] raft air gap

BambuLabDiscord, 2022-12-29:
Separation Distance – This is an important setting that defines the height of the air gap between the raft and your part. Typically, having a gap of at least 0.1mm will help keep the part connected to the raft, while still allowing for easy separation once the print is complete.

The Raft Air Gap setting is simply how large the gap is between the raft and the model itself. The larger this gap is, the easier it is to remove. It basically

lanewei120, 2023-01-03:
@BambuLabDiscord 
to better trace this issue, could you create this issue under https://github.com/bambulab/BambuStudio?

thanks

pepsiliu, 2023-01-03:
View  [https://github.com/bambulab/BambuStudio](BambuStudio)

closed 2023-01-03 by pepsiliu