bambulab/Bambu-Handy #54: [issue] No video from p1p in handy app

BambuLabDiscord, 2023-01-09:
after installing the camera in the p1p the bambu handy app shows “Connect failed” using the latest version of the iOS app and latest firmware on the printer.

pepsiliu, 2023-01-09:
Please add a ticket to us, we need logs from P1P.  After that you can reset the P1P as a temporary solution.

CCS86, 2023-01-17:
I just started a support ticket (US230117658001) for the same issue with my X1C.

Live video feed has been working perfectly in Handy until today. It still works fine in studio. Handy can issue commands to the printer, but not view video or browse media.

RyanEwen, 2023-01-24:
Try removing the printer from your account and then pairing it again. It won't lose any settings or its name. Had to do this for both of my P1Ps to get video to work after installing the cameras.

LibaoBambu, 2023-12-05:
Please reopen if this issue shows up again.

closed 2023-12-05 by LibaoBambu