bambulab/Bambu-Handy #58: Lamp state is not accurate in Handy

smford, 2023-01-11:
When the lamp is turned on when a job is printed from a computer, the status of the lamp does not get updated in Handy.

This will mean that the lamp will be on, but the handy app will say it is off.

pepsiliu, 2023-01-12:
Duplicated with #57

smford, 2023-01-12:
> Duplicated with #57

They appear very different from a user perspective.

#57 is about the P1P not turning off the lamp upon job completion - this could be in the gcode, or a setting in the P1P to timeout the lamp

whilst #58 is about changes in state to the lamp not being pushed to clients - or alternatively clients polling for the state of the lamp.

pepsiliu, 2023-01-12:
Yes, I have make a double check on the #57, It'll be transferred to P1P guys.

pepsiliu, 2023-01-16:
Fixed on P1P U1 hotfix. Will be released soon.

closed 2023-01-16 by pepsiliu