bambulab/Bambu-Handy #63: [Feature Request] IPAD compatibility

BambuLabDiscord, 2023-01-14:
Two requests. 
-Make the app Ipad ready
-Make the app a full solution becoming both Handy and Studio. for 99% of my prints i change maybe 5 to 10 parameters:
layer height
support yes/no and if yes its type (auto)
purge tower
color
paint colors (a must for any touch screen with a pen)
slowdown for overhangs
%infill
maybe one or two other
I believe a simplified version of BS could become an app and that would be a game changer. No other printer brand as it and theres no other printer such as the X1 right now!

Merlwynd, 2023-01-14:
What is meant by iPad ready? Handy works fine on my iPad…

reedy, 2023-11-20:
Numerous dupes of this... https://github.com/bambulab/Bambu-Handy/issues/130, https://github.com/bambulab/Bambu-Handy/issues/125, https://github.com/bambulab/Bambu-Handy/issues/134, https://github.com/bambulab/Bambu-Handy/issues/97

And this is kind of a dupe of https://github.com/bambulab/Bambu-Handy/issues/7