bambulab/Bambu-Handy #65: Android error notifications

Pchu24, 2023-01-16:
I only received print failed or print success notifications on my phone.

Only when I open up the app do I receive error notifications. I do not receive notifications when the AMS jams or fails to retract specifically. Are error notifications supposed to be sent through? 

Phone: Google Pixel 7
App Version: 1.0.8
Printer: X1C
Latest firmware on both the printer and AMS

Thank you!

pepsiliu, 2023-01-17:
Yes, printing works failed or success event will be sent through system (Android/iOS) notification currently.
The HMS (like the AMS jams or fails to retract specifically) is only a Handy event which only available by connecting the printer.

Pchu24, 2023-01-17:
I do not receive the error notifications when there are AMD jams or fails to retract. They do show up when I open up the app as messages in which I can click rety but no notifications.

pepsiliu, 2023-01-18:
> I do not receive the error notifications when there are AMD jams or fails to retract. They do show up when I open up the app as messages in which I can click rety but no notifications.

This is as expected. Only event which stop or pause the print works will be sent as system notification. So do the successful printing. Did the error from AMS stop the printing?

Pchu24, 2023-01-18:
Yes, the error stops printing and asks me to fix the error before retrying.

bjmckenz, 2023-03-03:
For me (x1c owner) I'm getting NO notifications as the first post said. Do I need to add a feature request so that they (all, not just the one above) are generated? (apologies if I'm not clear)

ercdvs, 2023-08-27:
Confirmed as a P1S owner.  I only get notifications on print finish and print cancel.  Nothing else comes through