bambulab/Bambu-Handy #66: "Connect Failed" When Trying to Start Live Video Feed

CCS86, 2023-01-18:
When I first setup BH, the live video feed worked perfectly. The brightness and smoothness of video are much better than they are in BS.

But, now when I launch BH, it connects to the printer, shows the correct machine state, temps, etc. But, when I click to start the video feed, after a few seconds of trying, I get a "connect failed" message. Live video still works fine in BS.

I have tried clearing all app data and reinstalling the app. The issue remains.

pepsiliu, 2023-01-19:
P1P or X1/X1C?

CCS86, 2023-01-19:
Sorry... X1C

Interesting wrinkle tonight. I tried again after leaving the printer, and on both mobile data and a remote wifi network, the video stream works.

I'll test again in the morning to see if being on the local wifi causes the same failure.

CCS86, 2023-01-19:
It is continuing to work on the local wifi network.

I have no idea what fixed the issue though, which is annoying!

pepsiliu, 2023-01-20:
I'm not sure, maybe the network is unstable or something whatever.
You can add ticket if the issue occurs again, so that we could check what's wrong from the logs.

Dexus, 2023-01-27:
Have the same Issue, my local network should be very good, but it's always needs a complete restart of the app, to reconnect.

guillaumemolter, 2023-02-07:
Yes same problem here. I have a pro-sumer grade network setup, no problems with any other apps or streaming. I also have another printer using octoprint and octopad app and the video is super fluid and seamless.

DapperDangus, 2023-02-09:
Pretty much my only gripe so far. It connects pretty easily after you start a print for a few hours even then after that I can maybe get it to connect 1/10 times. Not sure why it’s like that.

DapperDangus, 2023-02-09:
Pretty much my only gripe so far. It connects pretty easily after you start a print for a few hours even. Then after that I can maybe get it to connect 1/10 times. Not sure why it’s like that.

jsolarin, 2023-04-20:
Same problem as well. The X1C is on the same network as the IOS app. Bambu Studio on the desktop has no issue with the camera and downloading video files, but Handy says "Connect Failed" on video stream and browse videos shows "Device Connect failed, please try again[-12]".

The same ports are open from the phone to the printer that the desktop has so it can't be a firewall. Maybe a bad key or cert?

I've restarted the printer and iphone. Reinstalled the app. Factory reset the printer. Unbinded and rebinded the printer to the app. Still nothing.

pepsiliu, 2023-04-20:
> Same problem as well. The X1C is on the same network as the IOS app. Bambu Studio on the desktop has no issue with the camera and downloading video files, but Handy says "Connect Failed" on video [...]

It seems your Handy got some initialized exceptions. 
Please provide more informations:
1. Your iPhone model & version
2. The Handy version

jsolarin, 2023-04-20:
iPhone14pro on IOS 16.3.1

Handy app version 1.0.13

X1C firmware 01.04.01.00

jsolarin, 2023-04-23:
I connected to the printer from cellular data today, and out of no where, the video is working and I can see the time lapse videos. 

I wonder if my LAN was blocking a cert that it needed or something. I'll have to check my pi-hole logs.

Giran7, 2023-08-06:
Very strange, also could not get video to load, iphone/ipad/android phone, nothing. Went through network settings and phone settings couldn't get anything to work. Tried on cell, and it worked. Went back to wifi, force closed the app, and its working again.

Did you find anything that was blocking it? I guess I have a work around, going to go hotspot my phone to my ipad and see if that fixes it too.