bambulab/Bambu-Handy #67: [feature request] Push Notifications

markhellier, 2023-01-24:
Receive Push Notifications
- Print completion
- HMS Errors, spaghetti, run out of filament etc.

Currently you have to open the app to check if everything is still operating to see the errors.

RyanEwen, 2023-01-24:
I have the app installed on my phone as well as my wife's, and I was shocked to see that she gets notifications when prints are complete. I do not. Maybe notifications only work on one device at a time, and/or there are other issues making them not work fo everyone?

Guilouz, 2023-01-30:
Notifications are not working on iOS.

pepsiliu, 2023-01-31:
Notifications are only available on the last phone which had opened the handy currently.
We suggest u add a ticket about this kind of issue, we also need more logs from the printers.

closed 2023-05-17 by LibaoBambu