bambulab/Bambu-Handy #72: [Feature Request] Allow changing of filament via Bambu Handy

BambuLabDiscord, 2023-02-28:
I would like the ability to change filament spools from the AMS that are being used to print within Bambu Handy. 
One specific used case for this would be if I started a long print, left my house and the filament ran out. I should be able to choose a different filament color that is already loaded in the AMS with the same settings. I.e. generic black PLA when it runs out should allow me to switch to generic orange PLA or generic red PLA, and so on.￼ 
I should also be able to pause now/at next layer and do a filament change from the BH app with AMS. 
as of right now it only looks like auto switch filament woll work it the color is the same as well, wven if all other parameter are the same.

juliannikolovv, 2023-03-04:
Totally agree with you, at the moment my printer has stop because the filament run out, on slot 2 have the same filament even same settings and still can't make it to load it

fourtwizzy, 2023-03-05:
+1, such a great idea.

pepsiliu, 2023-03-06:
I have transferred this feature request to the printer's guys.

doudar, 2023-03-23:
I'd like to further elaborate on this one. I'd like the ability (at any time while printing) to be able to click on an new spool and have it automatically switch to that one (if the material parameters match what's already printing).