bambulab/Bambu-Handy #76: Android notifications do not honor printer name

AreYouRlyScared, 2023-03-05:
When receiving a push notification on android the custom printer name is not used, the original printer name is used. 

[image]

Name in app. 
[image]

Bambu Handy v1.0.11
Android v13 with Samsung OneUI v5.1
Samsung S23 Ultra

pepsiliu, 2023-03-06:
Thanks for your report. I have transferred this issue to push-service guys.

AreYouRlyScared, 2023-03-06:
Its also an issue on iOS, had my friend verify

pepsiliu, 2023-04-19:
The push-server had supported it.

closed 2023-04-19 by pepsiliu

AreYouRlyScared, 2023-04-25:
This is still broken and has not been fixed. I am using v1.0.14