bambulab/Bambu-Handy #78: [show the email associated with the bambu account somewhere in the app, and make it easy to change.] Bambu email

BambuLabDiscord, 2023-03-13:
A lot of the surveys/things sent out by Bambu ask for the email associated with my Bambu account. The problem is that I have at least 3 emails that I regulalry use when making accounts for various things and I sometimes forget which email I used for which app. It’ll be helpful to be able to go into “account” in bambu handy and see the email address used for that account. It would also be nice to be able to change the email on the app so I can move the account to a more convenient email. I could be the only one with this problem, but I feel like it should be an easyish fix.

pepsiliu, 2023-04-19:
It'll be released on next version.

pepsiliu, 2023-04-24:
fixed on 1.0.14

closed 2023-04-24 by pepsiliu