bambulab/Bambu-Handy #80: Support printer in LAN-only mode

m7r, 2023-03-16:
Hello,

in Bambu Studio it is possible to connect to printer in lan-only mode.

With the iOS app this is not possible. But it will help if you are at home or connected via VPN for a quick status check.

Thanks

enderjones, 2023-05-07:
+1 to adding this as a feature when on home WiFi.

AdrianGarside, 2023-05-21:
+1. With the constant bambu cloud outages, lan mode as the mitigation isn't completely when that means we lose all mobile monitoring and control via the handy app.

vordenken, 2023-06-30:
This really stops me from buying one of their printers. I want to locally control them. No cloud shenanigans and especially the certainty that the device I bought still fully works even if the company that created it goes out of business or just stops supporting the device.

mroek, 2023-08-30:
@LibaoBambu This is sorely needed, and technically very feasible. A comment as to whether this is on the roadmap would be very welcome.