bambulab/Bambu-Handy #86: Humidity indicator is in the app red

DokuKaefer, 2023-03-31:
Bambu Lab iPhone Version: 1.0.13

The humidity sensor is displayed in red in the iPhone app.
In the Bambu Studio and on the printer monitor it is green.

[image]

[image]

[image]

[image]

pepsiliu, 2023-03-31:
Yes, A bug on Handy.

brianiup, 2023-04-01:
same issue here always red

pepsiliu, 2023-04-24:
fixed on 1.0.14

closed 2023-04-24 by pepsiliu

Th199403, 2023-04-24:
@pepsiliu  when is the new app update for ios coming?

pepsiliu, 2023-04-24:
> @pepsiliu when is the new app update for ios coming?

 The Handy is pending apple and google approval, It'll be available for users after that.