bambulab/Bambu-Handy #88: [BUG] Using PAUSE during initial preparation of print will result in PAUSE getting "disabled"

bolsoncerrado, 2023-04-02:
OK, I was not sure where to fit this issue as I think it's related to the machine's firmware.

I use spools of filament that sometimes, when they run out, filament breaks inside the printer and it does NOT detect the filament runout so I find the machine printing on air. For this purpose, when using this specific brand, I tend to REPLACE the spool before it runs out. So this just happened:

I started a print. Realizing there was not much filament left, I hit PAUSE while the printer was preheating/bed leveling. 

It did NOTHING. THe print started.
Thinking it would pause after first layer inspection, I let it go. The PAUSE did not act.
I went to BambuStudio to try pressing PAUSE there. Nothing happened.
I went to HandyApp to try pressing PAUSE there. Nothing happened.

Last resort: I manually cut the filament from the spool hoping the runout sensor would kick in. NOTHING HAPPENED, NO PAUSE, NO NOTHING. 

I didn't want to stop the ongoing print so I manually pushed the filament until it resumed print WITHOUT PAUSE taking place!

So, to sum up, if you touch PAUSE on the screen while the printer is performing the startup routines, you will end up WITH NO PAUSE AND NO RUNOUT SENSOR for the rest of the print :(

pepsiliu, 2023-04-06:
Have transferred it to the printer's guys.
Initial preparation is unable to be paused, but can be stopped as I know.

bolsoncerrado, 2023-04-06:
Yes I know I could have stopped it but since I didnt want to, the BUG appeared hehe 

Thanks