bambulab/Bambu-Handy #94: [Issue] Infinate App Loading/Connecting to printer. requiring sign out and sign in every timr

BambuLabDiscord, 2023-04-14:
When opening app, printer view is not available and will be on loading screen continuously. i have to go yo settings, sign out, then sign back in yo gain functionality of the app and interact with the printer

pepsiliu, 2023-04-14:
@BambuLabDiscord Does it show the loading widget every time?

sam43434, 2023-04-16:
having the same issue

pepsiliu, 2023-04-17:
> having the same issue

Could u provide your account email and the time range when the issue occurs.

Blair1999, 2023-04-18:
Having the same issue here. Any time of day.

pepsiliu, 2023-04-19:
This issue will be fixed on next released version.

Blair1999, 2023-04-19:
> This issue will be fixed on next released version.

Awesome, thanks!

sam43434, 2023-04-19:
> This issue will be fixed on next released version.

is there any time frame for the next release, or just pending apple/ google approval

pepsiliu, 2023-04-20:
> > This issue will be fixed on next released version.
> 
> is there any time frame for the next release, or just pending apple/ google approval

A new version will be released as soon as this week

pepsiliu, 2023-04-24:
fixed on 1.0.14

closed 2023-04-24 by pepsiliu