bambulab/Bambu-Handy #97: [Feature Request] Pad OS and Watch OS

moebis, 2023-04-25:
The iPad version would look better if it wasn't a scaled version of the iPhone app. Some folks use it as a large display for controlling the printer (like me) and would be great if it fully utilized the screen real estate. Also even Octoprint has an app for Apple Watch so you can minimally control and watch prints from your watch.

LibaoBambu, 2023-05-17:
New version has been supported iPad.  And about Watch version, We will seriously consider supporting it on the watch, and perhaps release it in future version.

atl285, 2023-05-20:
Thanks for supporting iPad OS. Can you you add landscape mode to Bambu Handy. Especially for iPad it would be imporve the usability. But also on iPhone I would love to have a landscape mode useable.

LibaoBambu, 2023-05-22:
i will to think about it

moebis, 2023-06-03:
> New version has been supported iPad. And about Watch version, We will seriously consider supporting it on the watch, and perhaps release it in future version.

Thank you Libao it great! Watch and TV OS would be great too. I remember I used to monitor my prints with Octoprint on the Apple TV. Or if it was HomeKit compatible it could be part of the security and camera feeds all throughout Apple's ecosystem (which would include Apple TV and Watch OS too).

TheCableGuy99, 2023-06-09:
Just bought an iPad Pro and the Bambu Handy app is silly. Needs a resolution profile adding for it and also needs to support landscape mode. +1 for this please ASAP.