bambulab/Bambu-Handy #98: Only german language on v1.1.0

mainmind83, 2023-05-12:
On update to v1.1.0 only show german language
Uninstall and re-install (clear data and cache) with same result
Spanish language not showed...

[image]

BastR, 2023-05-13:
Same for French users, the app only show German.

When you go to “Me - Settings - Language”:

[image]

mainmind83, 2023-05-16:
V1.1.1

`This is a hotfix version. And it just fix the issue that the default language setting.
`

[image]

BastR, 2023-05-16:
For me the issues is not fixed, there no French nor Spanish

mainmind83, 2023-05-16:
@LibaoBambu can you review this bug? Regards

LibaoBambu, 2023-05-17:
@mainmind83 new version 1.1.1 has beed released in store.

closed 2023-05-17 by LibaoBambu

BastR, 2023-05-17:
Sorry to insist by v1.1.1 doesn’t solve the problem, some languages are missing:

https://github.com/bambulab/Bambu-Handy/assets/20129420/73279e23-0a33-4221-bf45-fb1563343744

LibaoBambu, 2023-05-17:
The other language translators are still working on it and will be released soon