bambulab/Bambu-Handy #103: [Feature Request] for better timelapse and video still integration

chuckkay, 2023-05-20:
Why would you put the browse timelapse videos hidden in the settings tab and not the "ME" section or at least in both.
Even better have the timelapse videos linked to the print history cards in the my history section..

Additionally when in the browse timelapse section, why not sort videos putting most recent first (not last)? it makes no sense, someone prints something chances are they go to browse timelapse section they are looking for the most recent timelapse so list videos in descending order by date (newest at the top) or at least add a sorting button...  also (YEAR MONTH ALL) not very helpful...unless segmented by actual year and actual month. but if you don't want to do that, at least add week and remove year (year isn't necessary ALL option is sufficient instead of year)

pepsiliu, 2023-05-22:
Thanks for your report. And let me answer your questions.

> Why would you put the browse timelapse videos hidden in the settings tab and not the "ME" section or at least in both.
Even better have the timelapse videos linked to the print history cards in the my history section..

The timelapse is strongly associated with the printer, so putting it to "ME" doesn't quit make sense, but we'll discuss this feature requirement and see how to deal with it to make the experience better.

> Additionally when in the browse timelapse section, why not sort videos putting most recent first (not last)? it makes no sense, someone prints something chances are they go to browse timelapse [...]

It's a known firmware bug, and will be fixed on next released version.

> also (YEAR MONTH ALL) not very helpful...unless segmented by actual year and actual month. but if you don't want to do that, at least add week and remove year (year isn't necessary ALL option is sufficient instead of year)

Yes, the tabs is not very helpful because our protocol does not support this feature very well, and we'll optimize the UI.

LibaoBambu, 2023-12-05:
Please reopen if there are any other requests.

closed 2023-12-05 by LibaoBambu