bambulab/Bambu-Handy #104: [Feature Request] Print Pre-sliced models directly from Handy or studio that you haven't print before

BambuLabDiscord, 2023-05-25:
Can we have a feature were we can print already sliced items via the Handy app or even the desktop app. There can even be a market that can live on the app where people can just select a print and hit send to printer. That or let us slice our own prints and after save them for later in the app for later printing. Kind of like the History section you can print again from but for prints that we have not yet printed. This would be amazing cause I have plenty of times where I slice models I want to print but dont want to print them at that time. Or give us the ability to select prints we have on our SD card that in the printer.

nickeeey, 2023-05-25:
i also want this feature and a cherry on the top would be a cloud slicing/phone slicing option with standard profiles we can upload from bambu studio to the printer/cloud

slatifi, 2023-06-03:
Would be amazing if we could just print whats on the SD from the app.

LibaoBambu, 2023-12-05:
Feature has been supported.

closed 2023-12-05 by LibaoBambu