bambulab/Bambu-Handy #108: [Issue] Bambu Handy Android App not updating tickets

BambuLabDiscord, 2023-06-16:
I've recently noticed that my app isn't showing my updated support ticket information. i have to use my email app to respond to support.

LibaoBambu, 2023-06-19:
Thanks for opening an issue!
Could you please describe the issue you are having?
Thanks!

3D-Pink-Girl, 2023-06-19:
> Thanks for opening an issue! Could you please describe the issue you are having? Thanks!

My support tickets aren't updating in the app. I've done clear cache and data and now uninstall/install. Still isn't showing latest updates that support have left. Any ideas?

3D-Pink-Girl, 2023-06-19:
Bambu Handy android app version 1.1.2

Previously it had no issues regarding the support tickets but ever since update it’s not showing new messages.

LibaoBambu, 2023-06-19:
[image]
Long press to provide your ticket number, I'll follow up on this issue.

LibaoBambu, 2023-06-19:
@3D-Pink-Girl  send ticket number to libao.jia@bambulab.com with email

3D-Pink-Girl, 2023-06-19:
I see that your version 1.2.0 actually fixed this issue in bug fixes

"Fixed the issue that some devices did not display ticket responses."

3D-Pink-Girl, 2023-06-19:
> @3D-Pink-Girl send ticket number to [libao.jia@bambulab.com](mailto:libao.jia@bambulab.com) with email

Yeah I've been responding to support by email as the app isn't updating the responses. I'll try 1.2.0 and let you know. Thanks again for your help

LibaoBambu, 2023-06-19:
The new version will be released tomorrow, and I hope it can solve your problem. If you have any tasks, please let me know. Thank you for your feedback.

3D-Pink-Girl, 2023-06-19:
> The new version will be released tomorrow, and I hope it can solve your problem. If you have any tasks, please let me know. Thank you for your feedback.

Oh that's awesome news. Thank you so much. I will report back on the results once I've updated to 1.2.0. Big thanks to you and the team here 🥇 😘 You're doing fab work

3D-Pink-Girl, 2023-06-21:
Hi, I still don't see the update yet 1.2.0. Are you still working on that build? Thanks x

3D-Pink-Girl, 2023-06-25:
I can't find the 1.2.0 update anywhere 😕

LibaoBambu, 2023-06-25:
please try again.   [image]

3D-Pink-Girl, 2023-06-25:
It came up this time lol. Thank you so much 😘

closed 2023-07-25 by LibaoBambu