bambulab/Bambu-Handy #112: [Issue] Important notifications not working-

BambuLabDiscord, 2023-06-30:
Neither on IOS nor on Android, although all notifications are allowed, will important notifications like "filament runout", "can't pull out filament" etc. not get displayed. 
Only notifications like "print cancelled" or "print finished" will get displayed.

drewzh, 2023-07-16:
Just adding my experience. Also experiencing the same issue on iOS. Notifications are enabled but do not work. I haven't had a single notification since getting the printer a few days ago.

psts, 2023-07-24:
Not working for me either. I got notifications for half a day some days ago and then back to not getting any. 
Have tried uninstalling and reinstalling the app.

psts, 2023-07-25:
I think I have figured out when it works and when it doesn't.

If you send a job from Bambu Studio, I get notifications.
If I start the job from the cache on the printer, I don't get notifications.

LibaoBambu, 2023-07-25:
@psts Because start the job from the cache on the printer does not go through our cloud service.

closed 2023-07-25 by LibaoBambu