bambulab/Bambu-Handy #121: [New feature] Handy app widgets for iOS

JakubComi, 2023-08-09:
Hello, I love the way you can monitor bambu lab printers. It would be perfect, if you could please add widgets for iOS, that way we can see faster what percentage is the print, or widget with camera! Thanks a lot for perfect printers and software! ;-)

LibaoBambu, 2023-08-11:
This feature has been recorded. We will process it as soon as other high-priority tasks are finished. Thanks for the feedback.

OzTamir, 2023-08-27:
+1000 For this feature, would be really helpful!

Schaumbee, 2023-10-26:
And a watch complication! I'd love seeing on my wrist the progress of a print

j-barot, 2024-02-03:
Live Activities would be awesome too !

GlennovitS, 2024-04-22:
Oh this would be so great to see! Maybe different versions, for different needs. Some with just info about the print (name, time left, time finished, printer name..) - and some with pictures from the camera 👍🏼 And some with a combination 🎉 

And maybe a small complication with time left on a selected printer + app for the Apple Watch to see the camera feeds 🎉

CoastJRA, 2024-05-17:
I would love this also make one for Android