bambulab/Bambu-Handy #125: [New feature] Vertical mode

antoan-boss, 2023-09-23:
Please add vertical mode to Bambu Handy app.

Thanks

F3000, 2023-10-03:
Useful for iPads

tkaravou, 2023-10-09:
I believe you mean horizontal mode. Currently the app is stuck in vertical mode making it nearly impossible to use when the iPad is in landscape orientation

F3000, 2023-10-11:
Yes you are right 👍

F3000, 2023-10-11:
But can't change title

antoan-boss, 2023-10-21:
I changed the title :) My bad..

BitScout, 2024-09-06:
Still nothing?

Tincho85, 2024-11-04:
+1 please

dotnvo, 2024-11-12:
@LibaoBambu  Could you comment if this is being considered for iOS/Android

JoomJ, 2024-11-23:
It would be an awesome feature on tabblets. I also have a Android tabblet with keyboard which makes forced switching a pain, also ergonomical as typing and rotating my head 90 degrees the whole time isn't helping my neck and shoulders at all. So im forced to use split screen (gives 2 smaller vertical apps on the horizontal screen) or other devices or flip te tabblet all the time. 

I think the biggest problem (or request) is in the "Device" tab. When i force the orientation on a OnePlus Tab (android) with an app (rotation control pro) to horizontal the other tab "Models" works ok and is scaling already (probable because there from the site itself), the "Me" tab works but needs some work, the topbar is huge. The "Device" tab doesnt respond well to rotation it shows the camerascreen large and doesn't respond to anything (swiping, scrolling, scaling). Hope to here what the plans are for this?