bambulab/Bambu-Handy #136: HMS_0700_2000_0003_0002 error in bambu handy cannot be cleared without turning P1S off

ThorsReign, 2023-10-10:
When printing using filament backup on the AMS, Bambu Handy will display a user friendly error to translate HMS_0700_2000_0003_0002 to "AMS1 Slot1 filament has run out and automatically switched to the slot with the same filament."

This message will not clear from Bambu Handy's device tab or the Machine Assistant portion of settings until the printer is reset, regardless of the number of prints you run since then, or adding new filament to the slot.  Irritating bug when the app is telling you that slot 1 ran out when you're actively printing from slot 1.

sodachen22, 2023-11-21:
@ThorsReign Does this problem always existed and what is the type of the printer, X1 or P1? If it is a ramdon issue, the reason could be the network is not stable enough. If this issue persists, please create a ticket and share the log.

BradysFidgets, 2024-06-09:
Not resolved. P1S/AMS, firmware 1.05.02. Having the exact same issue.

solsystemlabs, 2024-08-03:
Same here. Seems like this bug got reintroduced.

mroek, 2024-10-07:
I'm also seeing this on the X1C. Latest official firmware and Handy.

Landona24, 2025-02-25:
> I forgot I opened this. It's resolved as of the 1.04.02 firmware update

This issue is not resolved on the X1E either on 01.01.02.00.