bambulab/Bambu-Handy #137: [Issue] Not Receiving print error notifications

BambuLabDiscord, 2023-10-14:
Using the latest P1P firmware and latest app versions on both IOS and android I am not getting any "print paused with error" notications. I have uninstalled and reinstalled the handy app on both phones and still same issue.

BlackNet, 2023-10-14:
I concur.  I have been plagued with this for a few days now.  No error notices happen on the desktop or ios app.

porterble, 2023-10-18:
+1 

Latest Handy version. I get Print Success notifications but nothing else. Errors are displayed on screen when app is foregrounded.

RyanEwen, 2023-10-18:
Same here, for quite some time. I find out many hours after an error occurs and lose tons of print time as a result.

Hologos, 2023-10-28:
I haven't received any error notification ever on my phone, only success or when I canceled the print. If the filament runs put, I don't get the notification.

relax81, 2023-11-04:
I've the same problem. Received my printer a few days ago and I'm super happy with it, but I don't receive print errors in the mobile app or the smarthome integration.  

I get the printing done messages, but errors like "filament runout" or status messages like "Print paused" for planned pauses in the Slicer only show up in Bambu Studio for Windows - not on the Bambu Mobile App.

coyote1, 2023-11-05:
Same here, please add error notifications if filament on AMS stuck or any other errors.

edudg, 2023-11-07:
Same here, I've just submitted a support ticket and comment on the Play Store about it. It is a really really frustrating issue

sodachen22, 2023-11-21:
There will be no printer error notification message when the printing job is triggered from SD card. In other words, only printing job triggered from cloud has this notification message. Would you please help check the source of the printing job?

edudg, 2023-11-21:
There are **no** notifications about printing issues on the P1P,  **even when printing via cloud**.

> There will be no printer error notification message when the printing job is triggered from SD card. In other words, only printing job triggered from cloud has this notification message. Would you please help check the source of the printing job?

Hologos, 2023-11-21:
> There will be no printer error notification message when the printing job is triggered from SD card. In other words, only printing job triggered from cloud has this notification message. Would you please help check the source of the printing job?

No notification when printing using cloud or Banbu Handy print function from MakerWorld.

grrmisfit, 2024-01-07:
Can get notifications only if app is open and being looked at which is OK but need push notifications even if the screen is off so we know to look at check on it. This is not from SD card printing, only cloud. Printer is A1 running latest firmware and bambu studio is latest and bambu handy is latest

Hologos, 2024-01-27:
Seems like finally added that with the latest firmware.

[image]

reticivis-net, 2024-12-21:
> There will be no printer error notification message when the printing job is triggered from SD card. In other words, only printing job triggered from cloud has this notification message. Would you please help check the source of the printing job?

Is there any way to change this?