bambulab/Bambu-Handy #138: [Feature Request] Use Bambu Handy in Lan Mode

BambuLabDiscord, 2023-10-19:
Being able to start prints from the SD card or monitor and stop prints using Bambu Handy when the printer is in LAN mode would be very useful. I typically keep my printer in LAN mode As I don't have a need to leave it connected to the cloud all the time.

Starbase12, 2023-10-28:
That would be really great if we can use the mobile phone app without a cloud connection!

githubber4ever, 2023-12-23:
This feature world really be appreciated since the Bambu Studio can also access the printer in LAN mode.

vordenken, 2024-01-11:
Yes please add this feature. I like the idea of printing something on the go (without the need of slicing on my computer) but don't want to use all this cloud stuff. 
I am connected to my home network via VPN already so no need for cloud features.

For now I'm using lan mode and the homeassistant integration to check the webcam and stuff.

jwjames83, 2024-02-21:
Adding another "vote" for getting this implemented sooner rather than later.

aspdigital, 2024-05-17:
Another vote for eliminating the cloud from the connection between Bambu Handy to the printer.

denjohn, 2024-05-17:
Seems that the cloud is the connection to the cell. So what alternative is there when one wants to  view prints in progress ? The west doesn't trust this app thinking it's Chinese.  It's not.  Bambu is Japanese.  Please keep this cloud.

vordenken, 2024-05-17:
> Seems that the cloud is the connection to the cell. So what alternative is there when one wants to  view prints in progress ? The west doesn't trust this app thinking it's Chinese.  It's not.  Bambu is Japanese.  Please keep this cloud. 

It's not always black and white. You can keep the cloud option but just add lan only as well. Their printers/bambuStudio already can do this.

aspdigital, 2024-05-17:
> Seems that the cloud is the connection to the cell. So what alternative is there when one wants to view prints in progress ? The west doesn't trust this app thinking it's Chinese. It's not. Bambu is Japanese. Please keep this cloud.

If my printer and my smartphone are on the same network, there is no need for the machine to send info up to the cloud so the cloud can send it back down to my phone. This has nothing to do with who wrote the app or where their cloud is hosted. It is simply about how there is no need for the cloud step when everything is on the same network.

viktortrass, 2025-08-04:
come on guys include this function because we need it!!