bambulab/Bambu-Handy #143: Error -90 on video using cell service.

BlackNet, 2023-10-30:
Since the update it seems the video only works on wifi and not cellular service. It throws a -90 connect failed. Yet it will see the ams info, layers, messages etc. also my Voron works beautifully over the same cellular networks.

 [image]

DronBroks, 2023-11-09:
I confirm that the latest software update for iOS broke the ability to view videos in the application via cellular communication. The video appears after the 5th attempt to start viewing. In the previous version of the software, video worked perfectly.

etlevent, 2023-11-10:
Can you please provide the handy and the firmware versions?

pepsiliu, 2023-11-10:
Can you please provider the handy and the firmware versions.

DronBroks, 2023-11-10:
firmware versions  -  01.06.03.00
handy versions  -  2.2.0  iOs

pepsiliu, 2023-11-10:
> firmware versions - 01.06.03.00 handy versions - 2.2.0 iOs

Thanks. Did you add any ticket on this issue. It will be very helpful if we can get the logs from device.

BlackNet, 2023-11-10:
Handy: 2.2.0 on IOS
Firmware: 01.06.05.01 (latest version)
AMS: 00.00.06.40

BlackNet, 2023-11-10:
Ticket: US231109092001