bambulab/Bambu-Handy #144: Support Tickets do not show on Bambu Handy

jltendler, 2023-10-31:
Support ticket counts render and display as 0 in bambu handy, but I have a support ticket visible on the website.

jltendler, 2023-10-31:
Screenshots on the forum post here: https://forum.bambulab.com/t/bambu-handy-doesnt-show-support-tickets-in-app/33641

beibaochufa, 2023-11-17:
Do you still have this problem now?

jltendler, 2023-11-17:
Yes:
[image]
[image]

beibaochufa, 2023-11-27:
After our analysis, your Ticket is a MakerWorld Ticket, and you need to see it in the "MakerWorld Tickets" in the APP. Please find the following entrance to see if there is your ticket.
[image]

closed 2023-12-05 by LibaoBambu