bambulab/Bambu-Handy #146: [Feature Request] Have you considered USB 2 Ethernet support?

bolsoncerrado, 2023-11-20:
I know this is not related to Bambu Handy but wasn't sure where to post this.

Have you, specially for the P1 series, considered coding support in the firmware for a USB2Ethernet connectivity protocol so we can plug a converter in the P1 front panel USB port??

Specially useful in farm environments or places where walls are dimishing wifi speeds...

Idk?

Just my thoughts.

reedy, 2023-11-20:
https://github.com/bambulab/BambuStudio would be the other place, but I'm not sure either is necessarily better/worse...

Having an ethernet option, rather than relying on wifi would be good though :)

sodachen22, 2023-11-21:
I am sorry, the printer does not support USB2Ethernet protocol.

bolsoncerrado, 2023-11-21:
I know it probably does not. This is a Feature REQUEST :P