bambulab/Bambu-Handy #154: [Issue] Black camera feed since version 2.3.0 update

BambuLabDiscord, 2023-12-06:
Camera is not showing anything except a black screen since the most recent update 2.3.0 using wifi home network.

pepsiliu, 2023-12-07:
> Camera is not showing anything except a black screen since the most recent update 2.3.0 using wifi home network.

Please provider the system of the phone, it will be very helpful if there has a screenshot about this issue, thank you.

haxx0r07, 2023-12-14:
I have the same issue running 2.3.1 on iOS 17.2.

Once clicking the “play” button on the screen it just spins for a minute and then goes back to the play button.

renssies, 2023-12-18:
Same problem here, pressing play, loader spins, then goes back to play button. The view stays black. 

The problem occurs on cellular and on WiFi. Reinstalled the app, but didn’t fix it.

The app does have cellular and local network permission.

iOS Version: iOS 17.2/iPadOS 17.2
Device: iPhone 15 Pro, iPad Pro 11 Inch 2ng gen
Region: Europe
Printer: Bambu Lab X1C + 1 AMS
App version: 2.4.0

bjeanes, 2024-01-10:
I'm having the same issue, with an A1 (not mini).

faade21, 2024-01-18:
Same issue on X1C with latest Firmware
Bambu Handy 2.5
Samsung Galaxy S22 Ultra running Android 14, One UI 6.0

Video stream works once and then again after force closing and reopening Bambu Handy.

renssies, 2024-01-18:
> Same problem here, pressing play, loader spins, then goes back to play button. The view stays black.
> 
> The problem occurs on cellular and on WiFi. Reinstalled the app, but didn’t fix it.
> 
> The app does have cellular and local network permission.
> 
> iOS Version: iOS 17.2/iPadOS 17.2 
> Device: iPhone 15 Pro, iPad Pro 11 Inch 2nd gen 
> Region: Europe 
> Printer: Bambu Lab X1C + 1 AMS 
> App version: 2.4.0

After some unrelated issues I have reset the X1C, now the camera only works when on cellular and only works once, I need to force close and restart the app for the camera to work again on cellular.