bambulab/Bambu-Handy #161: No video or MicroSD access in Handy 2.4NO video or MicroSD access in Android

LJKraus, 2023-12-27:


pepsiliu, 2024-01-31:
-20011 means connection timeout with liveview/file-transport between mobile and printer. Please make the mobile & the printer on the same network and try again.

deftoner, 2025-03-30:
I found this post trying to find a solution for this issue. 

I have a P1S and a X1C. since the last version on Google Play, I can access content of SDCard. Everything else works (camera, printer config, filament, etc). 

From a computer I can see the SDCard without issues. In fact if I try to access the SD car from the phone and at the same time from the pc, pc will say that printer can handle more calls. so the phone is actually connecting to the SDCard. But latter will print the time out error. 

I tried all the common sense, Restar the phone, reinstall the app on phone. restart the printers, etc. 

From 3 pcs, access the printers works flawless. From my phone stopped working. 

Reddit says that is an issue on the latest version, that rolling back to Nov 2024 will fix it, but there is no official source for the APK, more than apk stores (that could be tampered with the apk of course).

deftoner, 2025-03-30:
Funny enough, while I was writing this, there was a new version in Google play, and that version fixed the access to the SD models, still have an error "File not found" on recordings. But I really dont care about those :)

Version is 3.0.2 (working)