bambulab/Bambu-Handy #162: [Feature request] Picture of the finished Print

resper, 2023-12-28:
For printers equipped with a camera, I would like to have the function that it takes a picture of the object at the end of the print before it lowers the printing plate. And then send this image with the completion message in the mobile phone app. This gives me an immediate overview of whether the print has turned out reasonably well. I don't usually use the timelaps function because I have no use for it and I don't constantly check my mobile phone to see if the print is OK. However, when the printer is finished, it often lowers the object so far that it is not immediately visible at first glance and you have to laboriously raise the printing plate first.

GlennovitS, 2024-04-22:
+1 from me! Would be really nice to see 👍🏼

LibaoBambu, 2024-04-22:
Requirement has been recorded and will be evaluated by corresponding colleagues.