bambulab/Bambu-Handy #171: [*Issue*] Failed to get AMS mapping table

BambuLabDiscord, 2024-01-29:
When starting a multicolour print file from the SD card of a P1 series with AMS from the Bambu Handy app, the printer will run into an error "Failed to get AMS mapping table [161305]". Pressing "Retry" does not fix the issue and the print must be aborted and started from Bambu Studio instead.

The issue appears only when starting the multicolour print on an SD card from the app. Starting the print from Bambu Studio (either as a new print, or stated from the Micro SD section of the printer) does not trigger this error.

bjwalker6, 2024-01-30:
Getting the same error even with single colour prints. Also happens even if I manually load the filament from the AMS via the app and then try to start the print. Printer pre-heats and then just sits there for about 5-10 minutes before displaying the error. Starting the same file printing from the SD card via Bambu studio or on the printers display itself works just fine also as you say.

pepsiliu, 2024-01-31:
hi, could any body provide more informations (Handy version & Firmware version & Filaments)? I have make a double check on printing with multicolor/singlecolor job in the sd card by the handy which works well.

jbellmore, 2024-02-01:
Mine has just started doing this in the last couple days as well. I've had mine for a few months at least and really the only thing that I have changed is that I just recently started using the "send" feature to send sliced models to the printer to start later.

The work flow that seems to cause it is:

- Slice the model in Bambu studio
- Send sliced plate to the printer (usually while it is printing something else)
- Start the print from browsing the SD card in the Bambu handy app

If I do the exact same model and instead of uploading then starting I just print directly from Bambu studio, seems to work fine.

I am on latest firmware (01.05.02.00) and latest version of Bambu handy app (2.5.0).

Hope that helps.

Thanks!

relax81, 2024-02-17:
I've had the same issue today.

The scenario was:
1) I send yesterday a multicolor print file through Bambu Studio to the printer while it was doing another print.   "Send", not "print". It was a hueforge with 2 color changes at specific layers.
2) This morning I opened the Bambu Handy App (Android) and started that print from SD Card. 
3) At the first color change I got the same error as people above - failed to get mapping table. A retry didn't help. 

Printer: P1S Version: 01.05.02.00(Latest version)
AMS: Version: 00.00.06.40(Latest version)
Bambu Studio (Windows): v1.8.4.51
Bambu Handy (Android): v2.6.1
SD Card: The original one it shipped with

Scotchy49, 2024-02-20:
Having the same issue repeatedly with my P1S with AMS.

Steps taken:

- Slice in Orca Slicer then send to printer's SD card through Orca Slicer UI.
- Go in Handy App and start the print of the file
- The file will print fine.

Some time/prints later (possibly swapping AMS filaments)
- Go in Handy App and start the print of the exact same file which printed fine before.
- Printer goes to "Heatbed pre-heating", stops for a while and then shows the AMS mapping table error.

The fix is to go back to Orca Slicer, re-slice the model and upload it again.

The issue never occured when printing straight from Orca Slicer.

It is extremely frustrating and this issue never happens with my X1C.

P1xellat3d, 2024-02-28:
Some further observations:

- Some users have reported that using a new SD card fixes the issue. Personally, I've tried a different SD card (not the one that the P1S came with, but a SanDisk one) and have still had the same issue.
- Whilst the "AMS mapping table" error is active, the printer holds the toolhead and Z axis in position. Unless you resolve the error by aborting the print, the stepper motors will continue to hold (and be supplied with current). In my case, this (over the course of a day) cause the stepper motors to become extremely hot and resulted in a separate error message: "**The resistance of Motor-A is abnormal, the motor may have failed.**". Leaving the printer off overnight to allow the stepper motors to cool resolved Motor-A issue. 
- As other users have noted above, the error is definitely tied to the SD card in some way, and not specific to the Handy app. Starting a print from the MicroSD section from Bambu Studio results in the same error.

davefp, 2024-02-28:
I'm seeing this too. It occurred for the first time yesterday and the only thing that seems to be different is that I've got a mix of materials in the AMS.

LibaoBambu, 2024-03-26:
Thank you for your feedback.  New version has been fixed this issue.

closed 2024-03-26 by LibaoBambu

rdbischoff, 2024-04-20:
what was the solution? i don't see any solution but it's "closed as completed"

LibaoBambu, 2024-04-22:
> what was the solution? i don't see any solution but it's "closed as completed"

This issue was fixed in App version2.7.0, Please reopen if this issue shows up again in new app version. Thanks.

ASAllon, 2024-07-03:
Hello,

I have a new p1s with latest app ams and printer firmware. The Error accord to again.... I've printed from handy app with two different matirales petg and pla and now it stack before first color change... Trying to pause and replay without nothing...

LibaoBambu, 2024-07-03:
> Hello,
> 
> I have a new p1s with latest app ams and printer firmware. The Error accord to again.... I've printed from handy app with two different matirales petg and pla and now it stack before first color change... Trying to pause and replay without nothing...

Please provide more information [https://bambulab.com/en-eu/my/support/tickets](url)

utiq, 2024-08-28:
@LibaoBambu it just happened to me, how it can be reproduced.
1. I sent a print from the Bambu Studio from **MacOS** (let's call it print XYZ)
2. When the print job finished, I sent started another print job in this way: In the Bambu Studio I went to `MicroSD Card` Model -> I selected the file with the model XYZ -> I clicked on print.
3. On  Bambu Studio **WINDOWS** I created a new print job with another 3D model and sent to the print (let's call it print ABC)
4. When the print job finished, I tried to do again step 2 and the print failed with the error *[Failed to get AMS mapping table* when the first color change triggered

I also noticed the files from the SD card got corrupted