bambulab/Bambu-Handy #174: "filament has run out" notification deos not clearfrom Bambu Handy

ElJeffe, 2024-02-08:
2 days ago, I got [this](https://wiki.bambulab.com/en/x1/troubleshooting/hmscode/0700_2000_0003_0002) notification on my app because the AMS switched filaments. I can however no clear the notification. It stays visible on the screen and in the Machine Assistant page of the Settings. There seem to be no way to confirm or clear the notification.
P1S version: 01.05.02.00
AMS: 00.00.06.40
BambuHandy (Android): 2.6.1

[image]
[image]

LibaoBambu, 2024-02-26:
Message should clear when Slot2 load a new filament. For the current firmware version you need to restart the printer to clear this hms message and firmware will fix this issue in the next version.

closed 2024-02-26 by LibaoBambu

RozeDoyanawa, 2024-05-09:
I have the same issue with my recently purchased P1S (firmware is the latest public)
HMS_0700_2000_0003_0002 is stuck and does not clear. It gets "replaced" when another slot runs out. But it does not clear.

M450NC, 2024-05-16:
Same issue here. AMS just purchased for P1S. Slot1 and Slot3 both ran out and this error persisted for ages. Had to reboot my printer at the plug 3 or 4 times to get the errors to go.

cyberhuman, 2024-07-07:
Why is this issue closed?
The notification persists on my X1E.

LibaoBambu, 2024-07-08:
> 为什么这个问题关闭了？ 我的 X1E 上仍然存在通知。

Please create a ticket to provide more info [https://bambulab.com/en-eu/my/support/tickets](url), Thanks.

bsaunder, 2024-07-21:
I'm having the same problem with my X1C on the latest firmware.

@LibaoBambu what firmware version allegedly resolved this issue?

DapperDangus, 2024-07-21:
@LibaoBambu same issue here. The notification cannot be dismissed until you reboot the printer.

houmark, 2024-07-21:
Same for me.

angusmaul, 2024-09-29:
Still not fixed, sooo annoying.

HDC67, 2024-10-29:
Still not fixed. I ran into this today with an up to date firmware on a brand new printer.

joethepeacock, 2024-11-02:
Still not fixed here - X1C

phillipse01, 2024-11-06:
@LibaoBambu this issue is still occurring - I have it on my a1 newest firmware today

RayBomb87, 2024-11-13:
Issue is still here x1c

TGS78, 2024-11-19:
So when will this be finally fixed?

bartoszpasiecznik, 2024-11-29:
P1S on latest firmware - Still not fixed, got the notification on Bambu Handy, P1S and Bambu Studio

thePunderWoman, 2024-12-04:
X1C on latest firmware - I have this issue right now. I added new filament to the slot, and the message persists on the app. No way to clear it.

TGS78, 2024-12-04:
> X1C on latest firmware - I have this issue right now. I added new filament to the slot, and the message persists on the app. No way to clear it.

actually you can only get rid of the notification by power cycling the printer.

adamurban, 2024-12-05:
Can we get this reopened? (Or make a new bug?)

i am still having the same issue on my X1C with the latest FW, even though I see this reported as fixed back in Feb?

@LibaoBambu can you confirm which firmware version this is resolved with? I am running 01.08.02.00 at the moment.

CodyPrince, 2025-09-13:
19 months later and this is still a problem. Why did they close this issue? maybe they can't fix it.