bambulab/Bambu-Handy #183: Request: Staging a file to history

lotsahelp, 2024-03-18:
There are times prints are running and I'd like to go ahead and upload the next print into history for easy printing from the app. This would allow me to get everything ready in Studio and a staff member to just print from the app the next time a printer becomes available. 

Thank you,

pepsiliu, 2024-07-16:
You can send a sliced task to the printer from the Studio. And start it via [Device Page]-[Settings (at thr ight top)]-[SD card]-[Models] by the Handy.

lotsahelp, 2024-07-16:
@pepsiliu I don't want to send it straight to a printer, as it would only be available on that printer. I want to send it straight to Handy's history so I can print it on the next available printer.

lotsahelp, 2024-07-16:
@pepsiliu How will that meet my needs for a private print farm?

pepsiliu, 2024-07-18:
@lotsahelp Hi, the Makerworld supports uploading private 3mf.  
For farm printing, there desn't have any better support, but upload to the history is definitely not a good solution.
Please describe in detail your needs and the solution you expect, and we will discuss it and put it into the feature-list.
Please open this issue if you think it is necessary. Thank you again.